NMGRL/pychron #1537: Data sorting in figures and tables

wcmcintosh, 2018-06-01:
Could the group node include a checkbox to presort dataset by group factor (typically ID) prior to grouping by that factor.  There are some scenarios where you would not want the data pre-sorted, but by default I would think that you would want this.  I don't think that Pychron pre-sorts in this way at present.

Question:  Can you remind me whether Analysis Groups preserve sorting of analysis lists when they are created? I'll follow up on this with some testing.

One other sorting issue: Some users prefer to see a data table of fusion data to be sorted by age, youngest analysis first.  Easy to do or not?  Does this create problems for type 3 data?

wcmcintosh, 2018-06-02:
Testing indicates that pychron currently does not sort the data used for figures and tables.

So desirable sorting options would be:
1) a choice to pre-sort prior to grouping
2) a choice to sort data groups by age

jirhiker, 2018-06-04:
Please make feature requests as granular as possible.  

Questions are best asked on slack. Plus it seems more of a feature request than a question. "Make analysis groups preserve order of analyses"

"One other sorting" this deserves is own github issue

jirhiker, 2018-06-04:
please you proper Markdown syntax

use task for a list of items. Do not manually number lists.  Buttons for number/bulleted/task lists are available just above

jirhiker, 2018-06-06:
seems like there is 3 levels of age sorting
1. individual analysis ages
2. subgroup ages
3. group ages

jirhiker, 2018-06-06:
sorting by group age cannot be accomplished without a significant redesign of how tables are constructed.

wcmcintosh, 2018-06-13:
when sort by age option is requested, group ages are still ordered by aliquot number. Data = 65520.

wcmcintosh, 2018-06-13:
Not sure what the current status of this issue is.  The main goal here was to be able to order individual analyses by age within a group of fusion analyses.  I wasn't thinking of ordering groups by age.  
Would ordering a group that contains both subgroups and fusion analyses be harder than ordering individual analyses?  
Step-heating data data with A=1, S>1 should not be reordered. The reason for labeling the issue as testing fail was testing of the current sorting option, which I now understand is not fully implemented.

closed 2019-09-11 by jirhiker