NMGRL/pychron #1538: Error during subgrouping in table

wcmcintosh, 2018-06-01:
active branch=None

active analyses=66265-01,66265-02,66265-03A,66265-03B,66265-04A,66265-04B,66265-05A,66265-05B,66265-06A,66265-06B,66265-07A,66265-07B,66265-08A,66265-08B,66265-09A,66265-09B,66265-10A,66265-10B,66265-11A,66265-11B,66265-12A,66265-12B,66265-13A,66265-13B,66265-14A,66265-14B,66265-15A,66265-15B,66265-16A,66265-16B,66265-17A,66265-17B,66265-18A,66265-18B,66265-19A,66265-19B,66265-20A,66265-20B,66265-21A,66265-21B,66265-22A,66265-22B,66265-23A,66265-23B,66265-24A,66265-24B,66265-25A,66265-25B,66265-26A,66265-26B,66265-27A,66265-27B,66266-01,66266-02A,66266-02B,66266-03A,66266-03B,66266-04A,66266-04B,66266-05A,66266-05B,66266-06A,66266-06B,66266-07A,66266-07B,66266-08A,66266-08B,66266-09A,66266-09B,66266-10A,66266-10B,66266-11A,66266-11B,66266-12A,66266-12B,66266-13A,66266-13B,66266-14A,66266-14B,66266-15A,66266-15B,66266-16A,66266-16B,66266-17A,66266-17B,66266-18A,66266-18B,66266-19A,66266-19B,66266-20A,66266-20B,66266-21A,66266-21B,66266-22A,66266-22B,66266-23A,66266-23B,66266-24A,66266-24B,66266-25A,66266-25B,66266-26A,66266-26B,66266-27A,66266-27B,66266-28A,66266-28B,66266-29A,66266-29B,66266-30A,66266-30B,66267-01,66267-02A,66267-02B,66267-03A,66267-03B,66267-04A,66267-04B,66267-05A,66267-05B,66267-06A,66267-06B,66267-07A,66267-07B,66267-08A,66267-08B,66267-09A,66267-09B,66267-10A,66267-10B,66267-11A,66267-11B,66267-12A,66267-12B,66267-13A,66267-13B,66267-14A,66267-14B,66267-15A,66267-15B,66267-16A,66267-16B,66267-17A,66267-17B,66267-18A,66267-19A,66267-19B,66267-20A,66267-20B,66267-21A,66267-21B,66267-22A,66267-22B,66267-23A,66267-23B,66267-24A,66267-24B,66267-25A,66267-25B,66267-26A,66267-26B,66267-27A,66267-27B,66267-28A,66267-28B,66267-29A,66267-29B,66267-30A,66267-30B,66268-01,66268-02,66269-01,66269-02A,66269-02B,66269-03A,66269-03B,66269-04A,66269-04B,66269-05A,66269-05B,66269-06A,66269-06B,66269-07A,66269-07B,66269-08A,66269-08B,66269-09A,66269-09B,66269-10A,66269-10B,66269-11,66269-12,66269-13,66269-14,66269-15,66269-16,66269-17,66269-18,66269-19,66269-20,66269-21,66269-22,66269-23,66269-24,66269-25,66269-26,66269-27,66269-28,66270-01,66270-02A,66270-02B,66270-03A,66270-03B,66270-04A,66270-04B,66270-05A,66270-05B,66270-06A,66270-06B,66270-07A,66270-07B,66270-08A,66270-08B,66270-09A,66270-09B,66270-10A,66270-10B,66270-11A,66270-11B,66270-12A,66270-12B,66270-13A,66270-13B,66270-14A,66270-14B,66270-15A,66270-15B,66270-16A,66270-16B,66270-17A,66270-17B,66270-18A,66270-18B,66270-19A,66270-19B,66270-20A,66270-20B,66270-21A,66270-21B,66270-22A,66270-22B,66270-23A,66270-23B,66270-24A,66270-24B,66270-25A,66270-25B,66270-26A,66270-26B,66270-27A,66270-27B,66270-28A,66270-28B,66270-29A,66270-29B,66270-30A,66270-30B,66271-01,66271-02,66271-03,66271-04,66271-05,66271-06,66271-07,66271-08,66271-09,66271-10,66271-11A,66271-11B,66271-11C,66271-11D,66271-12A,66271-12B,66271-12C,66271-12D,66271-13A,66271-13B,66271-13C,66271-13D,66271-14A,66271-14B,66271-14C,66271-14D,66271-15A,66271-15B,66271-15C,66271-15D,66271-16A,66271-16B,66271-16C,66271-16D,66271-17A,66271-17B,66271-17C,66271-17D,66271-18,66271-19,66271-20,66271-21,66271-22,66271-23,66271-24,66271-25,66271-26,66271-27,66271-28,66271-29,66271-30,66271-31,66271-32,66271-33,66271-34,66271-35,66271-36

description=""

Traceback
```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/trait_notifiers.py", line 519, in _dispatch_change_event
    self.dispatch( handler, *args )
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/trait_notifiers.py", line 482, in dispatch
    handler( *args )
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 729, in _preferred_kind_changd
    v = self._get_preferred_(obj.attr, obj.kind, obj.error_kind)
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 804, in _get_preferred_
    pa = self._calculate_integrated(attr)
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 344, in _calculate_integrated
    prs = ans[0].production_ratios
IndexError: list index out of range

```

wcmcintosh, 2018-06-01:
Still get error with subgrouping set to no grouping.  I'll try with SubGroup disabled

wcmcintosh, 2018-06-01:
No help from Subgroup disabled.  IR998 data runs fine in pipeline, there problem is data specific.

wcmcintosh, 2018-06-01:
Table works OK if skip tagged analyses removed.  IR998 had skipped analyses but ran OK.

jirhiker, 2018-06-04:
see #1533 the error is trigger by calculating an integrated age for an analysis group with no valid analyses

wcmcintosh, 2018-06-13:
OK I think.  Haven't encountered this error for over a week.  

Ready to close.

closed 2018-06-13 by jirhiker