NMGRL/pychron #1543: K/Ca mean

wcmcintosh, 2018-06-01:
Currently, in groups, mean K/Ca can only be weighted mean.  Once generalized mean are extended to groups as well as subgroups, the default mean type for K/Ca can be set to arithmetic mean, which is better for mixed plag and san

jirhiker, 2018-06-04:
what is the explicit request you are asking for. Im spending to much time trying to decipher these messages.

in its most simple from
"When I do X I want Y to happen"

I want to be able to right click an analysis and set its as invalid. After its set as invalid the analysis should be removed from the ideogram and the Unknowns Pane and the ideogram should be automatically replotted

jirhiker, 2018-06-06:
Addressed via subgrouping

wcmcintosh, 2018-06-13:
Arithmetic mean working well as mean type option for K/Ca.  
Still not sure what statistics are  best to use for somewhat wacky behaving K/Ca values.
Ready to close.

closed 2018-06-13 by jirhiker