NMGRL/pychron #1545: Error plotting table

wcmcintosh, 2018-06-02:
Error plotting table using IR1104 data (Panter feldspars, 3 spectra)
Gert same error with and without subgrouping

'''
PipelineEngine                          : 2018-06-01 18:31:54,089 DEBUG     (MainThread) pipeline run started
PipelineEngine                          : 2018-06-01 18:31:54,089 DEBUG     (MainThread) starting at node Review<ReviewNode> None
2018-06-01 18:31:59.115 python[8869:1503682] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
XLSXAnalysisTableWriter                 : 2018-06-01 18:31:59,118 DEBUG     (MainThread) saving table to /Users/mcintosh/Pychron_dev/data/tables/Untitled-069.xlsx
root                                    : 2018-06-01 18:31:59,558 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2018-06-01 18:31:59,558 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2018-06-01 18:31:59,558 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/xlsxwriter/worksheet.py", line 427, in write
    f = float(token)
root                                    : 2018-06-01 18:31:59,558 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/uncertainties/core.py", line 2631, in raise_error
    % (self.__class__, coercion_type))
root                                    : 2018-06-01 18:31:59,558 CRITICAL  (MainThread) TypeError: can't convert an affine function (<class 'uncertainties.core.AffineScalarFunc'>) to float; use x.nominal_value
root                                    : 2018-06-01 18:31:59,558 CRITICAL  (MainThread) During handling of the above exception, another exception occurred:
root                                    : 2018-06-01 18:31:59,559 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2018-06-01 18:31:59,559 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
root                                    : 2018-06-01 18:31:59,559 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
root                                    : 2018-06-01 18:31:59,559 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
root                                    : 2018-06-01 18:31:59,559 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 314, in resume
    self._resume_pipeline()
root                                    : 2018-06-01 18:31:59,559 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 498, in _resume_pipeline
    self._run('resume pipeline', 'resume_pipeline')
root                                    : 2018-06-01 18:31:59,559 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 480, in _run
    if not getattr(self.engine, func)():
root                                    : 2018-06-01 18:31:59,559 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 656, in resume_pipeline
    return self.run_pipeline(state=self.state)
root                                    : 2018-06-01 18:31:59,560 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 751, in run_pipeline
    node.run(state)
root                                    : 2018-06-01 18:31:59,560 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/persist.py", line 256, in run
    writer.build(state.groups, options=self.options)
root                                    : 2018-06-01 18:31:59,560 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 75, in build
    self._make_human_unknowns(unknowns)
root                                    : 2018-06-01 18:31:59,560 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 402, in _make_human_unknowns
    self._make_sheet(unks, 'Unknowns')
root                                    : 2018-06-01 18:31:59,560 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 506, in _make_sheet
    cum=ag.cumulative_ar39(i) if ag else '')
root                                    : 2018-06-01 18:31:59,560 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 773, in _make_analysis
    sh.write(row, j + 1, txt, fmt)
root                                    : 2018-06-01 18:31:59,560 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/xlsxwriter/worksheet.py", line 65, in cell_wrapper
    return method(self, *args, **kwargs)
root                                    : 2018-06-01 18:31:59,560 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/xlsxwriter/worksheet.py", line 432, in write
    raise TypeError("Unsupported type %s in write()" % type(token))
root                                    : 2018-06-01 18:31:59,560 CRITICAL  (MainThread) TypeError: Unsupported type <class 'uncertainties.core.AffineScalarFunc'> in write()
root                                    : 2018-06-01 18:31:59,561 CRITICAL  (MainThread) ============ End Exception ==========

'''

jirhiker, 2018-06-04:
again is backticks not apostrophes. 

use the Preview view to make sure you are formatting correctly

jirhiker, 2018-06-05:
There is no project or repository called IR1104

wcmcintosh, 2018-06-13:
Error corrected.  
Ready too close.

closed 2018-06-13 by jirhiker