NMGRL/pychron #1563: SD for N=1

wcmcintosh, 2018-06-13:
Currently, arithmetic mean for an n=1 dataset (all but one analysis omitted), the mean correctly shows up as the value of the one valid point, but SD =#NUM!.  IN n=1 cases like this, could the n=1 error be used for SD, rather than  SD =#NUM!?[](url)

wcmcintosh, 2018-06-13:
Related to #1566

wcmcintosh, 2018-06-13:
Table pipeline seems be be handling n=1 data much better.  Group mean correctly using uncertainty for single analysis. One modification (see attached file) would be to not include total integrated age for n=1 groups.  Also to use group mean types, not subgroup mean types where n=1.  E.g. K/Ca mean type. Only affects label, not values.

[3.xlsx](https://github.com/NMGRL/pychron/files/2100332/3.xlsx)

jirhiker, 2018-06-14:
this are individual requests. Please separate your requests

closed 2018-06-14 by jirhiker