NMGRL/pychron #1565: total integrated age problem - not show-stopper

wcmcintosh, 2018-06-13:
See per attached table.  Total integrated age only appropriate for single-aliquot groups or subgroups.  Currently included with fusion data where aliquots > 1. 

[2.xlsx](https://github.com/NMGRL/pychron/files/2096331/2.xlsx)

jirhiker, 2018-06-13:
The user has control of whether the integrated age is display as part of the summary rows

wcmcintosh, 2018-06-13:
"The user has control of whether the integrated age is display as part of the summary rows"

I haven't tested your change yet, but in a Type 3 data set, wouldn't it make sense to show integrated age for single-aliquot step-heat data, but not formulti-aliquot fusion data (inappropriate statistic)

jirhiker, 2018-06-13:
simply stating that the user has control.

"I haven't tested your change yet, but in a Type 3 data set, wouldn't it make sense to show integrated age for single-aliquot step-heat data, but not formulti-aliquot fusion data (inappropriate statistic)" I agree in this case but this is another example where we fundamentally disagree. 

Its not up to the developer to define whats a sensible use of statistics, its up to the user to understand what they are asking of the application.

wcmcintosh, 2018-06-13:
Total integrated seems to be being applied correctly to step heating data, and not to fusion data.  All working well. Ready to close.

closed 2018-06-13 by jirhiker