NMGRL/pychron #1574: Error in final node of Table pipeline

wcmcintosh, 2018-06-18:
active branch=None

active analyses=63457-01A,63457-01B,63457-01C,63457-01D,63457-01E,63457-01F,63457-02A,63457-02B,63457-02C,63457-02D,63457-02E,63457-02F,63457-02G,63457-03A,63457-03B,63457-03C,63457-03D,63457-04A,63457-04B,63457-04C,63457-04D,63457-05A,63457-05B,63457-05C,63457-05D,63458-01A,63458-01B,63458-01C,63458-01D,63458-02,63458-03,63458-04,63458-05,63458-06,63458-07,63458-08,63458-09,63458-10,63461-01A,63461-01B,63461-01C,63461-01D,63461-02A,63461-02B,63461-02C,63461-02D,63461-03A,63461-03B,63461-03C,63461-03D,63461-04A,63461-04B,63461-04C,63461-04D,63461-05A,63461-05B,63461-05C,63461-05D,63461-05E,63461-05F,63462-01A,63462-01B,63462-02A,63462-02B,63462-03A,63462-03B,63462-04A,63462-04B,63462-05A,63462-05B,63465-01A,63465-01B,63465-01C,63465-01D,63465-02A,63465-02B,63465-02C,63465-02D,63465-03A,63465-03B,63465-03C,63465-03D,63465-04A,63465-04B,63465-04C,63465-04D,63465-05A,63465-05B,63465-05C,63465-05D,63465-06A,63465-06B,63465-07A,63465-07B,63465-08A,63465-08B,63465-09A,63465-09B,63465-09C,63465-10A,63465-10B,63465-11A,63465-11B,63465-12A,63465-12B,63465-13A,63465-13B,63465-13C,63467-01A,63467-01B,63467-01C,63467-01D,63467-02A,63467-02B,63467-02C,63467-02D,63467-03A,63467-03B,63467-03C,63467-03D,63467-03E,63467-03F,63467-04A,63467-04B,63467-04C,63467-04D,63467-04E,63467-04F,63467-05A,63467-05B,63467-05C,63467-05D,63467-05E,63467-05F,63468-01A,63468-01B,63468-01C,63468-01D,63468-02,63468-03,63468-04,63468-05,63468-06,63470-01A,63470-01B,63470-01C,63470-01D,63470-02A,63470-02B,63470-02C,63470-02D,63470-03A,63470-03B,63470-03C,63470-03D,63470-04A,63470-04B,63470-04C,63470-04D,63470-05A,63470-05B,63470-05C,63470-05D,63470-06A,63470-06B,63470-07A,63470-07B,63470-08A,63470-08B,63470-09A,63470-09B,63470-10A,63470-10B,63470-11A,63470-11B,63472-01A,63472-01B,63472-01C,63472-01D,63472-02A,63472-02B,63472-02C,63472-02D,63472-03A,63472-03B,63472-03C,63472-03D,63472-03E,63472-03F,63472-04A,63472-04B,63472-04C,63472-04D,63472-04E,63472-04F,63472-05A,63472-05B,63472-05C,63472-05D,63472-05E,63472-05F,63473-01A,63473-01B,63473-01C,63473-01D,63473-02,63473-03,63473-04,63473-05,63473-06,63473-07,63473-08,63473-09,63473-10,63473-11,64101-01B,64101-01C,64101-01D,64101-01E,64101-01F,64101-02A,64101-02B,64101-03A,64101-03B,64101-04A,64101-04B,64101-05B,64101-06A,64101-06B,64101-07A,64101-07B,64101-08A,64101-08B,64101-09A,64101-09B,64101-10A,64101-10B,64101-11A,64101-11B,64101-12A,64101-12B,64101-13A,64101-13B,64102-01A,64102-01B,64102-01C,64102-02A,64102-02B,64102-02C,64102-03A,64102-03B,64102-03C,64102-04A,64102-04B,64102-04C,64102-05A,64102-05B,64102-05C,64102-06A,64102-06B,64102-06C,64102-07A,64102-07B,64102-07C,64102-08A,64102-08C,64102-09A,64102-09B,64102-09C,64102-10A,64102-10B,64102-10C,64102-11A,64102-11B,64102-11C,64102-12A,64102-12B,64102-12C,64103-01B,64103-01C,64103-01D,64103-01E,64103-01F,64103-02A,64103-02B,64103-03A,64103-03B,64103-04A,64103-04B,64103-05A,64103-05B,64103-06A,64103-06B,64103-07A,64103-07B,64103-08A,64103-08B,64103-09A,64103-09B,64103-10A,64103-10B,64103-11A,64103-11B,64103-12A,64103-12B,64103-13A,64103-13B,64104-01A,64104-01B,64104-01C,64104-01D,64104-01E,64104-01F,64104-02A,64104-02B,64104-03A,64104-03B,64104-04A,64104-04B,64104-05A,64104-05B,64104-06A,64104-06B,64104-07A,64104-07B,64104-08A,64104-08B,64104-09A,64104-09B,64104-10A,64104-10B,64104-11A,64104-11B,64105-01A,64105-01B,64105-02A,64105-02B,64105-03A,64105-03B,64105-04A,64105-04B,64105-05A,64105-05B,64105-06A,64105-06B,64105-07A,64105-07B,64105-08A,64105-08B,64105-09A,64105-09B,64105-10A,64105-10B,64105-11A,64105-11B,64105-12A,64105-12B,64105-13A,64105-13B,64105-14A,64105-14B,64107-01A,64107-01B,64107-02A,64107-02B,64107-03A,64107-03B,64107-04A,64107-04B,64107-05A,64107-05B,64107-06A,64107-06B,64107-07A,64107-07B,64107-08A,64107-08B,64107-09A,64107-09B,64107-10A,64107-10B,64107-11A,64107-11B,64107-12A,64107-12B,64107-13A,64107-13B,64109-01A,64109-01B,64109-01C,64109-02A,64109-02B,64109-02C,64109-03A,64109-03B,64109-03C,64109-04B,64109-04C,64109-05A,64109-05B,64109-05C,64109-05D,64109-05E,64109-05F,64109-05G,64109-06A,64109-06B,64109-06C,64109-06D,64109-06E,64110-01A,64110-01B,64110-02A,64110-02B,64110-03A,64110-03B,64110-04A,64110-04B,64110-05A,64110-05B,64110-06A,64110-06B,64110-07A,64110-07B,64110-08A,64110-08B,64110-09A,64110-09B,64110-10A,64110-10B,64110-11A,64110-11B,64110-11C,64110-12A,64110-12B,64110-12C,64110-13A,64110-13B,64110-13C,64111-01A,64111-01B,64111-01C,64111-02A,64111-02B,64111-02C,64111-03A,64111-03B,64111-03C,64111-04A,64111-04B,64111-04C,64111-05A,64111-05B,64111-05C,64111-06A,64111-06B,64111-06C,64111-07A,64111-07B,64111-07C,64111-08A,64111-08B,64111-08C,64111-09A,64111-09B,64111-09C,64111-10A,64111-10B,64111-10C,64111-11A,64111-11B,64111-11C,64111-12A,64111-12B,64111-12C

description="Error in final node of Table pipeline - all Toba sanidine analyses
Will try subset to test data dependence"

Traceback
```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 318, in resume
    self._resume_pipeline()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 502, in _resume_pipeline
    self._run('resume pipeline', 'resume_pipeline')
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 484, in _run
    if not getattr(self.engine, func)():
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 659, in resume_pipeline
    return self.run_pipeline(state=self.state)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 757, in run_pipeline
    node.run(state)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/persist.py", line 256, in run
    writer.build(state.groups, options=self.options)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 98, in build
    self._make_summary_sheet(unknowns)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 477, in _make_summary_sheet
    txt = self._get_txt(ug, ci)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 1100, in _get_txt
    v = func(item, attr)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/util.py", line 82, in value
    x = getattr(x, k)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/has_traits.py", line 896, in decorator
    self.__dict__[ name ] = result = function( self )
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 182, in _get_isochron_age
    a = self.calculate_isochron_age()
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 424, in calculate_isochron_age
    args = self.get_isochron_data()
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 419, in get_isochron_data
    return calculate_isochron(ans, self.isochron_age_error_kind, exclude=exclude)
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/argon_calculations.py", line 93, in calculate_isochron
    yint = ufloat(reg.get_intercept(), reg.get_intercept_error())
  File "/Users/mcintosh/Programming/git/pychron/pychron/core/regression/new_york_regressor.py", line 140, in get_intercept_error
    return self.get_intercept_variance() ** 0.5
  File "/Users/mcintosh/Programming/git/pychron/pychron/core/regression/new_york_regressor.py", line 249, in get_intercept_variance
    self.get_slope_variance()
  File "/Users/mcintosh/Programming/git/pychron/pychron/core/regression/new_york_regressor.py", line 296, in get_slope_variance
    var_b = a / dVdb ** 2
ZeroDivisionError: division by zero

```

wcmcintosh, 2018-06-18:
Yes, data dependent.  OK for first analysis in data group.

jirhiker, 2018-06-18:
tagging as skip does not mean don't include in table.

The meaning of skip is defined in preferences.

wcmcintosh, 2018-06-18:
True.  But prides were set to exclude skips from everything but machine table.
[4.pdf](https://github.com/NMGRL/pychron/files/2112261/4.pdf)

wcmcintosh, 2018-06-18:
error after last update

```
DVC                                     : 2018-06-18 11:48:12,756 DEBUG     (MainThread) Make analysis time, total: 37.66935706138611, n: 623, average: 0.06046445756241751
PipelineEngine                          : 2018-06-18 11:48:12,810 DEBUG     (MainThread) 00: Unknowns<UnknownNode> Runtime: 0.0534
2018-06-18 11:48:15.936 python[6058:12869259] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
PipelineEngine                          : 2018-06-18 11:48:16,035 DEBUG     (MainThread) 01: Grouping<GroupingNode> Runtime: 0.0986
2018-06-18 11:48:19.642 python[6058:12869259] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
PipelineEngine                          : 2018-06-18 11:48:19,743 DEBUG     (MainThread) 02: SubGroup<SubGroupingNode> Runtime: 0.1000
not auto configure
root                                    : 2018-06-18 11:48:19,861 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2018-06-18 11:48:19,861 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2018-06-18 11:48:19,861 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
root                                    : 2018-06-18 11:48:19,861 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
root                                    : 2018-06-18 11:48:19,862 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
root                                    : 2018-06-18 11:48:19,862 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 315, in run
    self._run_pipeline()
root                                    : 2018-06-18 11:48:19,862 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 505, in _run_pipeline
    self._run('run pipeline', 'run_pipeline')
root                                    : 2018-06-18 11:48:19,862 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 484, in _run
    if not getattr(self.engine, func)():
root                                    : 2018-06-18 11:48:19,862 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 757, in run_pipeline
    node.run(state)
root                                    : 2018-06-18 11:48:19,862 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/group_age.py", line 35, in run
    editor.make_groups()
root                                    : 2018-06-18 11:48:19,862 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/editors/group_age_editor.py", line 174, in make_groups
    bind_preference(self, 'skip_meaning', 'pychron.pipeline.skip_meaning')
root                                    : 2018-06-18 11:48:19,862 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/apptools/preferences/preference_binding.py", line 172, in bind_preference
    return PreferenceBinding(**traits)
root                                    : 2018-06-18 11:48:19,863 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/apptools/preferences/preference_binding.py", line 42, in __init__
    self._set_trait(notify=False)
root                                    : 2018-06-18 11:48:19,863 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/apptools/preferences/preference_binding.py", line 139, in _set_trait
    trait_value = self._get_value(self.trait_name, value)
root                                    : 2018-06-18 11:48:19,863 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/apptools/preferences/preference_binding.py", line 116, in _get_value
    return handler.validate(self, trait_name, value)
root                                    : 2018-06-18 11:48:19,863 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/trait_types.py", line 507, in validate
    self.error( object, name, value )
root                                    : 2018-06-18 11:48:19,863 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/trait_handlers.py", line 173, in error
    value )
root                                    : 2018-06-18 11:48:19,863 CRITICAL  (MainThread) traits.trait_errors.TraitError: The 'skip_meaning' trait of a PreferenceBinding instance must be a boolean, but a value of 'Human Table,Ideogram,Isochron,Series,Spectrum' <class 'str'> was specified.
root                                    : 2018-06-18 11:48:19,863 CRITICAL  (MainThread) ============ End Exception ==========
ignore exception

```

wcmcintosh, 2018-06-18:
Making progress.  No error related to isochron calculations, and skip data treatment looks good.  I did get the following K2O related error in the Excel node.  This problem is definitely data dependent, and I will investigate which data is responsible.

```
2018-06-18 17:38:22.784 python[15063:13315292] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
XLSXAnalysisTableWriter                 : 2018-06-18 17:38:22,794 DEBUG     (MainThread) saving table to /Users/mcintosh/Pychron_dev/data/tables/Untitled-130.xlsx
root                                    : 2018-06-18 17:38:23,799 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2018-06-18 17:38:23,799 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2018-06-18 17:38:23,799 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
root                                    : 2018-06-18 17:38:23,799 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
root                                    : 2018-06-18 17:38:23,799 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
root                                    : 2018-06-18 17:38:23,800 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 318, in resume
    self._resume_pipeline()
root                                    : 2018-06-18 17:38:23,800 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 502, in _resume_pipeline
ignore exception
    self._run('resume pipeline', 'resume_pipeline')
root                                    : 2018-06-18 17:38:23,800 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 484, in _run
    if not getattr(self.engine, func)():
root                                    : 2018-06-18 17:38:23,800 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 659, in resume_pipeline
    return self.run_pipeline(state=self.state)
root                                    : 2018-06-18 17:38:23,800 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 757, in run_pipeline
    node.run(state)
root                                    : 2018-06-18 17:38:23,800 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/persist.py", line 256, in run
    writer.build(state.run_groups, options=self.options)
root                                    : 2018-06-18 17:38:23,800 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 74, in build
    unknowns = self._make_human_unknowns(unknowns)
root                                    : 2018-06-18 17:38:23,801 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 431, in _make_human_unknowns
    return self._make_sheet(unks, 'Unknowns')
root                                    : 2018-06-18 17:38:23,801 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 576, in _make_sheet
    cum=a.cumulative_ar39(ii) if a else '')
root                                    : 2018-06-18 17:38:23,801 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 892, in _make_analysis
    txt = self._get_txt(item, c)
root                                    : 2018-06-18 17:38:23,801 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 1100, in _get_txt
    v = func(item, attr)
root                                    : 2018-06-18 17:38:23,801 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/util.py", line 82, in value
    x = getattr(x, k)
root                                    : 2018-06-18 17:38:23,801 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/processing/arar_age.py", line 141, in display_k2o
    return '' if not self.weight else self.k2o
root                                    : 2018-06-18 17:38:23,801 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/processing/arar_age.py", line 134, in k2o
    moles_k = moles_39k * klambda / (k40_k*nominal_value(self.j))
root                                    : 2018-06-18 17:38:23,801 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/uncertainties/core.py", line 658, in f_with_affine_output
    f_nominal_value = f(*args_values, **kwargs)
root                                    : 2018-06-18 17:38:23,802 CRITICAL  (MainThread) ZeroDivisionError: float division by zero
root                                    : 2018-06-18 17:38:23,802 CRITICAL  (MainThread) ============ End Exception ==========
```

wcmcintosh, 2018-06-19:
This error was probably user error.  I can't seem to reproduce it, using the same data.  Table pipeline running error free.  Ready to close.

wcmcintosh, 2018-06-19:
I may have sorted the data in some funny way.

closed 2018-06-19 by jirhiker