NMGRL/pychron #1581: R

wcmcintosh, 2018-06-20:
Attempted to recall 63458-03, analysis not recalled, caused following error message

```
Pipeline Data Processing                : 2018-06-20 10:07:59,722 DEBUG     (MainThread) recall 63458-03 <pychron.database.records.isotope_record.DVCIsotopeRecordView object at 0x132c7fa58>
DVC                                     : 2018-06-20 10:07:59,723 DEBUG     (MainThread) sync repository Toba. exists=True
Inspection                              : 2018-06-20 10:07:59,729 DEBUG     (MainThread) pull called by sync_repo. parent call=<pychron.dvc.dvc.DVC object at 0x1a34fee9e8> repo.pull(use_progress=use_progress)
GitRepoManager                          : 2018-06-20 10:07:59,729 DEBUG     (MainThread) pulling master from origin
GitRepoManager                          : 2018-06-20 10:07:59,730 DEBUG     (MainThread) pulling from url: https://github.com/NMGRLData/Toba.git
git.cmd                                 : 2018-06-20 10:07:59,731 DEBUG     (MainThread) Popen(['git', 'fetch', 'origin'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba, universal_newlines=False, shell=None)
GitRepoManager                          : 2018-06-20 10:08:00,167 DEBUG     (MainThread) fetch complete
git.cmd                                 : 2018-06-20 10:08:00,167 DEBUG     (MainThread) Popen(['git', 'merge', 'FETCH_HEAD'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba, universal_newlines=False, shell=None)
GitRepoManager                          : 2018-06-20 10:08:00,219 DEBUG     (MainThread) pull complete
ISO                                     : 2018-06-20 10:08:00,290 WARNING   (MainThread)  - Failed loading peakcenter. error='int' object is not iterable
DVC                                     : 2018-06-20 10:08:00,290 DEBUG     (MainThread) make analysis exception
DVC                                     : 2018-06-20 10:08:00,290 DEBUG     (MainThread) Traceback (most recent call last):
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 798, in func
    reload=reload, *args)
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 1390, in _make_record
    pname, prod = productions[a.irradiation][a.irradiation_level]
KeyError: 'E'

DVC                                     : 2018-06-20 10:08:00,295 DEBUG     (MainThread) Make analysis time, total: 0.5722851753234863, n: 1, average: 0.5722851753234863
Pipeline Data Processing                : 2018-06-20 10:08:00,296 WARNING   (MainThread) failed making records

```

jirhiker, 2018-06-20:
your metadata repository is missing a production for NM-274 E

wcmcintosh, 2018-06-20:
How can I fix this? When I go to lab number entry, everything looks OK for NM-274E production ratios.  I toggled the selection and saved to database, but still can't recall 63458 data.  When I try to recall 63458 data, a window briefly opens, but closes before I can read it.

wcmcintosh, 2018-06-20:
I restarted pychron and all seems OK.

wcmcintosh, 2018-06-20:
Thanks for figuring out that the problem was related to the missing production ratio.   No longer a problem.  Perhaps a warning should be displayed to the user if this happens in the future.   Ready to close.

closed 2018-06-20 by jirhiker