NMGRL/pychron #1582: Subgroup means not displayed when there is only one subgroup

wcmcintosh, 2018-06-20:
Subgroup means not displayed when there is only one subgroup in a group.  Grouping is fine and %39 correctly calculated.

For groups with more than one subgroup, all subgroup means are correctly displayed.

See attached Excel spreadsheet.

\[121.xlsx](https://github.com/NMGRL/pychron/files/2120242/121.xlsx)

wcmcintosh, 2018-06-22:
Single subgroups correctly displaying means and errors.  All OK except for very minor formatting issue - an extra table-wide cell border is added between single subgroups and the subgroup means.  Does not occur for multiple subgroups.  Attached excel spreadsheet shows extra line.

[4.xlsx](https://github.com/NMGRL/pychron/files/2128628/4.xlsx)

jirhiker, 2018-06-22:
fixed the previous specific issue. Likely broke for other types of data

wcmcintosh, 2018-06-23:
Formatting problem all fixed.  Cell borders all OK I think.  I'll try more testing of single aliquot datasets.

closed 2019-09-11 by jirhiker