NMGRL/pychron #1588: IC factor options configuration

wcmcintosh, 2018-06-21:
When running an IC factor pipeline, the IC factor options configuration remembers the name of the last used configuration, but the actual details in some cases (e.g. after program restart) do not match the configuration until the user toggles to another configuration and back again.  This behavior seems different than that of most other configuration windows.  I'm wondering why.

jirhiker, 2018-06-21:
Its the problem that its very difficult to intuit whether the user wants to use the defaults or the previous configuration

jirhiker, 2018-06-21:
If you except no defaults for the IC factor pipeline template then it would probably work much better

wcmcintosh, 2018-06-21:
The current model is kind of half of each, previous name but default parameters.

jirhiker, 2018-06-21:
yep

wcmcintosh, 2018-06-21:
Did you make a change to no defaults, use persisted value?  I haven't tested this yet but can.

jirhiker, 2018-06-22:
it uses the defaults if the "Default" configuration is selected otherwise it should use the persisted values

wcmcintosh, 2018-06-22:
Working well with persisted values.  Ready to close.

closed 2018-06-22 by jirhiker