NMGRL/pychron #1589: Subgroup ideogram problem

wcmcintosh, 2018-06-21:
Data 63457-01 to -05.  Subgroup ideogram is correctly subgrouping this data, but the resultant ideogram is plotting all steps rather than the subgroup means.

jirhiker, 2018-06-21:
I think its fixed for 

I tested on 
[SubG1, SubG2, SubG3]
and 
[SubG1, A1,A2,A2]
type data

wcmcintosh, 2018-06-21:
Almost 100%.  All subgroup mean parameters plotting correctly except 39Ar(fa), which is plotting as zero.  Values in table tab look perfect, but not making it to the graph.  see attached file
[1.pdf](https://github.com/NMGRL/pychron/files/2125676/1.pdf)

wcmcintosh, 2018-06-21:
I should also note that other than the 39Ar(fa) problem, all figure and table values are agreeing exactly, which is really great.

wcmcintosh, 2018-06-22:
Mean 39Ar(fa) values still plotting as zero. Data 63457-01 to -05.

wcmcintosh, 2018-06-22:
All fixed.  Ready to close.

closed 2018-06-22 by jirhiker