NMGRL/pychron #1592: subgroup mean error type

wcmcintosh, 2018-06-22:
subgroup mean for plateau ages reporting SEM error, regardless of what is selected configuration.

SEM/MSWD, SEM, SD all give SEM value

wcmcintosh, 2018-06-22:
Spectrum pipeline handles error choice correctly

wcmcintosh, 2018-06-22:
Likewise, group plateau using arithmetic mean ± SD instead of selection.  All a consequence of fixing the single group format issue?

wcmcintosh, 2018-06-22:
Possibly related issue in Spectrum pipeline,  Spectrum options: spectrum  allows selection of error type for weighted-mean age.  Error for weighted-mean age is SEM regardless of selection.  Error type for plateau correctly responds to choice.

wcmcintosh, 2018-06-22:
Subgroup plateau ages now reporting SEM/mswd errors correctly.
Subgroup weighted mean ages still reporting SEM errors.

jirhiker, 2018-06-22:
"Subgroup weighted mean ages still reporting SEM errors." could not reproduce

wcmcintosh, 2018-06-22:
Same for spectra - plateau age errors are SEM/MSWD
Weighted mean errors SEM.

jirhiker, 2018-06-22:
The behavior of the spectrum is not necessarily related to the behavior of the subgroups

wcmcintosh, 2018-06-22:
I wasn't sure but thought it a possibility.

wcmcintosh, 2018-06-23:
No longer a problem.  I think that all error kind selections are now correct. Ready to close.

closed 2018-06-26 by jirhiker