NMGRL/pychron #1594: Excel final node error

wcmcintosh, 2018-06-23:
Working with full Toba dataset.  I got this error if I change any of the group error kinds to SD.  Runs fine if defaults are not modified

```
2018-06-23 13:53:08.491 python[13018:17922734] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
PipelineEngine                          : 2018-06-23 13:53:08,555 DEBUG     (MainThread) 01: Grouping<GroupingNode> Runtime: 0.0638
2018-06-23 13:53:10.775 python[13018:17922734] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
PipelineEngine                          : 2018-06-23 13:53:10,846 DEBUG     (MainThread) 02: SubGroup<SubGroupingNode> Runtime: 0.0707
not auto configure
not auto configure
PipelineEngine                          : 2018-06-23 13:53:19,667 DEBUG     (MainThread) 03: Group Age<AnalysisTableNode> Runtime: 8.8109
PipelineEngine                          : 2018-06-23 13:53:19,669 DEBUG     (MainThread) 04: Review<ReviewNode> Runtime: 0.0023
PipelineEngine                          : 2018-06-23 13:53:19,670 DEBUG     (MainThread) pipeline vetoed by Review<ReviewNode>
Pipeline Data Processing                : 2018-06-23 13:53:19,678 DEBUG     (MainThread) false run pipeline run_pipeline
Pipeline Data Processing                : 2018-06-23 13:53:19,851 DEBUG     (MainThread) run pipeline finished
Pipeline Data Processing                : 2018-06-23 13:53:33,777 DEBUG     (MainThread) resume pipeline started
pychronmeta                             : 2018-06-23 13:53:33,778 DEBUG     (MainThread) force create new session 112497924496
PipelineEngine                          : 2018-06-23 13:53:33,778 DEBUG     (MainThread) pipeline run started
PipelineEngine                          : 2018-06-23 13:53:33,778 DEBUG     (MainThread) starting at node Review<ReviewNode> None
2018-06-23 13:53:38.408 python[13018:17922734] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
XLSXAnalysisTableWriter                 : 2018-06-23 13:53:38,413 DEBUG     (MainThread) saving table to /Users/mcintosh/Pychron_dev/data/tables/Untitled-173.xlsx
root                                    : 2018-06-23 13:53:39,256 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2018-06-23 13:53:39,256 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2018-06-23 13:53:39,257 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
root                                    : 2018-06-23 13:53:39,257 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
root                                    : 2018-06-23 13:53:39,257 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
root                                    : 2018-06-23 13:53:39,257 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 318, in resume
    self._resume_pipeline()
root                                    : 2018-06-23 13:53:39,257 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 502, in _resume_pipeline
    self._run('resume pipeline', 'resume_pipeline')
root                                    : 2018-06-23 13:53:39,257 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 484, in _run
    if not getattr(self.engine, func)():
root                                    : 2018-06-23 13:53:39,257 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 659, in resume_pipeline
    return self.run_pipeline(state=self.state)
root                                    : 2018-06-23 13:53:39,257 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 757, in run_pipeline
    node.run(state)
root                                    : 2018-06-23 13:53:39,258 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/persist.py", line 256, in run
    writer.build(state.run_groups, options=self.options)
root                                    : 2018-06-23 13:53:39,258 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 74, in build
    unknowns = self._make_human_unknowns(unknowns)
root                                    : 2018-06-23 13:53:39,258 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 431, in _make_human_unknowns
    return self._make_sheet(unks, 'Unknowns')
root                                    : 2018-06-23 13:53:39,258 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 576, in _make_sheet
    cum=a.cumulative_ar39(ii) if a else '')
root                                    : 2018-06-23 13:53:39,258 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 887, in _make_analysis
    txt = self._get_txt(item, c)
root                                    : 2018-06-23 13:53:39,258 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 1095, in _get_txt
    v = func(item, attr)
root                                    : 2018-06-23 13:53:39,258 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/util.py", line 82, in value
    x = getattr(x, k)
root                                    : 2018-06-23 13:53:39,258 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/processing/arar_age.py", line 141, in display_k2o
    return '' if not self.weight else self.k2o
root                                    : 2018-06-23 13:53:39,258 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/processing/arar_age.py", line 134, in k2o
    moles_k = moles_39k * klambda / (k40_k*nominal_value(self.j))
root                                    : 2018-06-23 13:53:39,259 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/uncertainties/core.py", line 658, in f_with_affine_output
    f_nominal_value = f(*args_values, **kwargs)
root                                    : 2018-06-23 13:53:39,259 CRITICAL  (MainThread) ZeroDivisionError: float division by zero
root                                    : 2018-06-23 13:53:39,259 CRITICAL  (MainThread) ============ End Exception ==========
ignore exception

```

wcmcintosh, 2018-06-23:
Still get same error if any error kinds changed from default to SD

jirhiker, 2018-06-23:
you did not pull the latest changes

wcmcintosh, 2018-06-23:
Runs without error after changing group means to SD. Ready to close.

closed 2018-06-23 by jirhiker