NMGRL/pychron #1595: K/Ca ideogram scaling

wcmcintosh, 2018-06-23:
I'm trying to understand what controls the max and min in the K/Ca scale on ideograms.  Many datasets yield a sensible scale (e.g. 63457-05), others (e.g. 63467-04C,D,E) have a narrow range of values and unknowns (in this case all between 50 and 150) , but set a too-wide scale (in this case -4294 to +5254).  I am wondering how the algorithm works.

jirhiker, 2018-06-23:
the scaling takes into account error bars

wcmcintosh, 2018-06-23:
Even with error bars, there appears to be some inconsistency in scaling.  For example, 63467-04C,D,E values and errors all fall within 50-150, whereas the scaling is 30x larger than it needs to be.

wcmcintosh, 2018-06-24:
K/Ca scaling on ideograms works exactly right.  Spectrum scaling algorithm must be operating differently.

jirhiker, 2018-06-26:
[63467-003.pdf](https://github.com/NMGRL/pychron/files/2135288/63467-003.pdf)

wcmcintosh, 2018-06-29:
Additional insight:  Behavior is history dependent.  In spectrum, analyses in some cases inherit the large K/Ca scale from the previously plotted analyses.  Restarting pychron and replotting yields correct scaling.

jirhiker, 2018-07-17:
This is why I hate dealing with scaling. Its very difficult for the program to intuit what the user wants, e.i. use previous limits, calculate new limits, use user defining limits, etc.  

An excessive amount of time has been dedicated to this problem already and this issue will likely not be revisited until other issues dictate a need to work on spectrum graphing.

closed 2018-07-17 by jirhiker