NMGRL/pychron #1599: CSV Ideogram pipeline error

wcmcintosh, 2018-06-26:
got following error running file attached below, which I think matches format detailed in recent slack message

```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 315, in run
    self._run_pipeline()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 505, in _run_pipeline
    self._run('run pipeline', 'run_pipeline')
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 484, in _run
    if not getattr(self.engine, func)():
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 757, in run_pipeline
    node.run(state)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/data.py", line 156, in run
    unks = self._load_analyses()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/data.py", line 169, in _load_analyses
    return self._get_items_from_file(par)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/data.py", line 185, in _get_items_from_file
    return tuple(gen())
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/data.py", line 178, in gen
    age_err=float(d['age_err']),
KeyError: 'age_err'
[Toba ages.txt](https://github.com/NMGRL/pychron/files/2138791/Toba.ages.txt)

```

usgshelix, 2018-06-26:
`age_error` in the csv should be `age_err`

```
age_err=float(d['age_err']),
KeyError: 'age_err'
```

wcmcintosh, 2018-07-26:
[Toba ages.txt](https://github.com/NMGRL/pychron/files/2230001/Toba.ages.txt)

I tried again with the attached file, which has correct "age_err" column heading, and got the following error

Do you have working model file that I can see?  That might help me determine the error of my ways.

```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 315, in run
    self._run_pipeline()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 505, in _run_pipeline
    self._run('run pipeline', 'run_pipeline')
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 484, in _run
    if not getattr(self.engine, func)():
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 757, in run_pipeline
    node.run(state)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/data.py", line 157, in run
    unks = self._load_analyses()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/data.py", line 171, in _load_analyses
    return self._get_items_from_file(par)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/data.py", line 189, in _get_items_from_file
    return tuple(gen())
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/data.py", line 179, in gen
    for d in parser.values():
AttributeError: 'CSVColumnParser' object has no attribute 'values'

```

wcmcintosh, 2018-07-27:
Got further, but new error in last node.

Also, even if CSV ideogram pipeline reset of cleared, pipeline only asks you to choose file the first time pipeline is run.  To access a new file, user must toggle out of CSV ideogram and back again.

```
PipelineEngine                          : 2018-07-26 18:11:17,418 DEBUG     (MainThread) pipeline runtime 25.160504817962646
PipelineEngine                          : 2018-07-26 18:11:17,419 DEBUG     (MainThread) pipeline post run started
PipelineEngine                          : 2018-07-26 18:11:17,419 DEBUG     (MainThread) post run node 00: CSV Data
PipelineEngine                          : 2018-07-26 18:11:17,419 DEBUG     (MainThread) post run node 01: Ideogram
PipelineEngine                          : 2018-07-26 18:11:17,419 DEBUG     (MainThread) pipeline post run finished
root                                    : 2018-07-26 18:11:17,424 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2018-07-26 18:11:17,424 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2018-07-26 18:11:17,425 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
root                                    : 2018-07-26 18:11:17,425 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
root                                    : 2018-07-26 18:11:17,425 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
root                                    : 2018-07-26 18:11:17,425 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 315, in run
    self._run_pipeline()
root                                    : 2018-07-26 18:11:17,425 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 505, in _run_pipeline
    self._run('run pipeline', 'run_pipeline')
root                                    : 2018-07-26 18:11:17,425 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 484, in _run
    if not getattr(self.engine, func)():
root                                    : 2018-07-26 18:11:17,425 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 781, in run_pipeline
    self.post_run(state)
root                                    : 2018-07-26 18:11:17,425 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 798, in post_run
    self.post_run_refresh(state)
root                                    : 2018-07-26 18:11:17,425 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 816, in post_run_refresh
    self._update_repository_status()
root                                    : 2018-07-26 18:11:17,426 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 865, in _update_repository_status
    r.update()
root                                    : 2018-07-26 18:11:17,426 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/tasks/repo_task.py", line 62, in update
    a, b = ahead_behind(p, fetch=fetch)
root                                    : 2018-07-26 18:11:17,426 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/git_archive/utils.py", line 117, in ahead_behind
    repo = get_repo(repo)
root                                    : 2018-07-26 18:11:17,426 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/git_archive/utils.py", line 140, in get_repo
    repo = Repo(repo)
root                                    : 2018-07-26 18:11:17,426 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/git/repo/base.py", line 161, in __init__
    raise InvalidGitRepositoryError(epath)
root                                    : 2018-07-26 18:11:17,426 CRITICAL  (MainThread) git.exc.InvalidGitRepositoryError: /Users/mcintosh/Pychron_dev/data/.dvc/repositories

```

closed 2019-07-21 by jirhiker