NMGRL/pychron #1605: Force plateau not working and in strange place

nelsiverson, 2018-06-29:
When I try to use the force plateau it does not work. Does not change anything in the graph. Also could this be put in the 'plateau' tab, seems a bit out of place in 'groups'.

[Screen Shot 2018-06-29 at 3.52.10 PM.pdf](https://github.com/NMGRL/pychron/files/2151055/Screen.Shot.2018-06-29.at.3.52.10.PM.pdf)

wcmcintosh, 2018-06-30:
My two cents on this issue:  The current weighted mean age, as an alternative mean age where plateau criteria are not met, is a far better way to treat imperfect data than "forced plateau".  I suggest that forced plateau be dropped from pychron.  One advantage to using omit and weighted mean age instead of forced plateau is the the mean age becomes more reproducible without having to reenter "forced plateau" steps in the group configuration each time the pipeline is run, and that tables and figures will consistently report the same value.  The weighted mean age is numerically identical to "forced plateau".  Forced plateau" is an oxymoron at best, and we can leave it behind without shedding a tear.

jirhiker, 2018-07-17:
I don't think getting rid of "forced plateau" will be acceptable. 

I agree "forced plateau" is an oxymoron. Pychron uses the term "Fixed plateau" but "plateau" probably shouldn't be in the name. 

The Fixed Plateau settings are in "Groups" so that the user has finer control of which steps to calculate an age for when dealing with multiple groups. e.g Group 1 fix A-D , Group 2 fix B-E etc

" One advantage to using omit and weighted mean age instead of forced plateau is the the mean age becomes more reproducible without having to reenter "forced plateau" steps in the group configuration each time the pipeline is run, and that tables and figures will consistently report the same value"  True but tagging analyses just to remove them from the calculations is cumbersome and likely an unacceptable workflow. 

Conclusion
Both omit/weighted-mean and "fixed plateau" are necessary

wcmcintosh, 2018-07-23:
Perhaps cumbersome, but persistent, which is a big advantage over keeping track of A-D, B-E, etc., and reentering the selections each time.

closed 2018-07-23 by jirhiker