NMGRL/pychron #1606: Question: Data table organization- Intercept and Corrected intercept switched?

nelsiverson, 2018-06-29:
From Heizler- "As with the mass spec AUF file, I cannot use the corrected intensities in Pychron table to calculate the reported age.  One thing I notice  is that the corrected intensity of 40Ar is greater than the intercept value. This is okay for things like 37Ar, 39Ar, and 36Ar where there are positive multipliers (decay, IC), but not expected for 40Ar." 

Our thought is that the two columns are switched in the table.  Attached is a screen shot of the table and the recall window. It seems like the intercept should be 340 and the corrected value should be 338. Is this possible?

[Screen Shot 2018-06-29 at 4.27.51 PM.pdf](https://github.com/NMGRL/pychron/files/2151115/Screen.Shot.2018-06-29.at.4.27.51.PM.pdf)

[Screen Shot 2018-06-29 at 4.32.18 PM.pdf](https://github.com/NMGRL/pychron/files/2151116/Screen.Shot.2018-06-29.at.4.32.18.PM.pdf)

wcmcintosh, 2018-06-30:
I agree that intercept and corrected values are switched in the table.

wcmcintosh, 2018-07-23:
Fixed.  Ready to close.

closed 2018-07-23 by jirhiker