NMGRL/pychron #1717: Integrated age not calculated/displayed correctly

mjz1983, 2019-03-05:
Integrated age for TinajaDikes (1 and 2) as displayed on spectrum is not correct. Compare pdf to excel data table.
[TinajaDikes.xlsx](https://github.com/NMGRL/pychron/files/2932105/TinajaDikes.xlsx)
[TinajaDikesSpectra_1column.pdf](https://github.com/NMGRL/pychron/files/2932106/TinajaDikesSpectra_1column.pdf)

jirhiker, 2019-03-05:
likely has to do with how the weighting for integrated age calculation is being done and that you do not have it set the same for the figure and the table.

mjz1983, 2019-03-05:
Seems like a pretty major error not explaind by weighting factors. Check out the age differences (~23 Ma vs. >100Ma)

Z

Matthew Zimmerer
Field Geologist II
NM Bureau of Geology
801 Leroy Pl
Socorro, NM 87801
matthew.zimmerer@nmt.edu
575.835.5114
www.mattzphotography.com

jirhiker, 2019-03-05:
k. you might be right

jirhiker, 2019-03-05:
k. the problem is definitely related to weighting. When you make the spectrum make sure to set the weighting to volume or variance. There currently is a bug in the integrated age calculation if the weighing is set to `---`

closed 2019-04-16 by jirhiker