NMGRL/pychron #1718: Spectra steps plotting with different colors

mjz1983, 2019-03-05:
When plotting a gray spectrum, some steps appear lt gray and some appear mod. gray. Or maybe it is a difference is the opaque setting. Seems reproducible regardless of color.
[image]
[image]

jirhiker, 2019-03-05:
this is a consequence of a WiscAr request but its also a little bizarre. 

in the Spectrum options under "Integrated Age" there is a option called "Include omitted"  Deselect that

mjz1983, 2019-03-05:
Toggling "include omitted" has no impact. Spectrum still plots with different opaque settings regardless of whether the option is toggled.

jirhiker, 2019-03-05:
sorry wrong option. 
Try "Dim No Plateau"

closed 2019-04-16 by jirhiker