NMGRL/pychron #1749: Add fit Flux to history on unknowns and monitors

nmgrluser, 2019-05-01:
Suggestion to add when the flux is fit to the history of the unknown and when the flux is calculated to the monitors.

Justification for this requested enhancement is when l looking at "older" data that was reduced months ago vs. plotting the data now there are minor discrepancies that could be explained if the J curve was refit and applied to the data. A history of each time the flux was calculated and applied to the unknowns would potentially help diagnose this inconsistency. Perhaps more of a problem when multiple projects from multiple PIs have samples in the same irradiation tray.

Maybe this is addressed in data version control and MZ is unaware of it.

jirhiker, 2019-05-01:
added a history pane to the Labnumber entry window. Shows when flux has been fit for a given irradiation. In the future it will show the history specific for a given level. (There was a bug in the flux fitting history message that prevents getting the history for a given level)

nmgrluser, 2019-06-17:
Not working. See NM303L. Neither unknowns or monitors have J flux added to history. Data worked up on 6/17/19.

jirhiker, 2019-06-17:
Flux history is not displayed in the Recall History. It is displayed in the Entry/Irradiation history pane

closed 2021-02-09 by jirhiker