NMGRL/pychron #1788: problems plotting spectra from Jan-pychron

mjz1983, 2019-07-19:
When plotting figures in pyExp on Jan, the plots do not show numerical values. Also, there are problems with ± and sigma symbols.

Z
[image]

jirhiker, 2019-07-19:
try changing the font. This is a computer specific font issue that is unresolved

mjz1983, 2019-07-19:
Problem with font is consistent with all available fonts. Also, for clarification, the font issue is when viewing it in pyChron, not in Adobe Illustrator

jirhiker, 2019-07-19:
Copy the viewing in pychron. Thats what I thought

When you change the font does the displayed font actually change or is it always that mono-spaced 
"typewriter-y" font

mjz1983, 2019-07-19:
Nope. It actually changes the font.

jirhiker, 2019-07-19:
on thing to try is clearing the font cache.  

rename to $HOME/.enthought/kiva/fontList.cache to $HOME/.enthought/kiva/~fontList.cache

closed 2019-12-02 by mjz1983