NMGRL/pychron #1793: Tagged omit for a plateau or weighted mean of step-heated sanidine 66144

nmgrluser, 2019-07-20:
active branch=develop

active analyses=00-22:Pl

description=""

Traceback
```
Traceback (most recent call last):
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/tasks/task.py", line 249, in set_tag
    self.dvc.tag_items(tag, items, note)
  File "/Users/argonlab2/.pychron.1/updates/pychron/dvc/dvc.py", line 1200, in tag_items
    self.debug('setting {} tag= {}'.format(it.record_id, tag))
AttributeError: 'NoneType' object has no attribute 'record_id'

```

jirhiker, 2019-07-21:
related to #1602

jirhiker, 2019-07-22:
cannot tag subgroup points

closed 2019-07-22 by jirhiker