NMGRL/pychron #1821: Error when plotting only analyses without probability curve

nmgrluser, 2019-09-11:
active branch=develop

active analyses=

description=""

Traceback
```
Traceback (most recent call last):
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/traits/trait_notifiers.py", line 519, in _dispatch_change_event
    self.dispatch( handler, *args )
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/traits/trait_notifiers.py", line 614, in dispatch
    handler( *args )
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/traitsui/ui.py", line 825, in _evaluate_when
    self._do_evaluate_when(at_init=False)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/traitsui/ui.py", line 839, in _do_evaluate_when
    self._evaluate_condition(self._visible, 'visible', at_init)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/traitsui/ui.py", line 884, in _evaluate_condition
    context = self._get_context( self.context )
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/traitsui/ui.py", line 805, in _get_context
    context2 = value.trait_get()
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/traits/has_traits.py", line 1484, in trait_get
    value = getattr( self, name, Missing )
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/traits/has_traits.py", line 896, in decorator
    self.__dict__[ name ] = result = function( self )
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/plot/editors/graph_editor.py", line 105, in _get_component
    comp = self._component_factory()
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/plot/editors/figure_editor.py", line 96, in _component_factory
    self.figure_container.model_changed()
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/plot/figure_container.py", line 62, in model_changed
    self.refresh(clear=clear)
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/plot/figure_container.py", line 47, in refresh
    comp.add(p.make_graph())
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/plot/panels/figure_panel.py", line 120, in make_graph
    fig.plot(plots, legend)
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/plot/plotter/ideogram.py", line 210, in plot
    self._rebuild_ideo(selection)
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/plot/plotter/ideogram.py", line 704, in _rebuild_ideo
    lp = plot.plots['Current-{}'.format(gid)][0]
KeyError: 'Current-1'

```

mjz1983, 2019-09-11:
Error occurs when plotting from CSV file or from database.

jirhiker, 2019-09-11:
plotting ideogram's with no probability curve was never anticipated or implemented

closed 2019-09-11 by jirhiker