NMGRL/pychron #1822: labelling analyses not working

mjz1983, 2019-09-11:
When toggling "Labels" for Analyses on CSV ideogram, the plot does not update. Analysis points remain un-labelled with RUN ID name. Only a problem with CSV ideogram. Works fine for ideogram from database.

jirhiker, 2019-09-11:
You need to specify the attribute to use to label the points. 
[image]

mjz1983, 2019-11-25:
Labelling points not working, regardless of whether using predefined labels or attempting to creating unique labels. Label box is tagged. See screenshoot. Maybe user is missing something???

[image]

jirhiker, 2019-11-25:
you need to turn labeling on per aux plot

mjz1983, 2019-11-26:
OK.

closed 2019-11-26 by mjz1983