NMGRL/pychron #1826: add plot figure to CSV ideogram

mjz1983, 2019-09-13:
For the CSV ideogram, there are currently two options for inputing data, either from a text file or adding the ages in manually. When using a text file, CSV ideogram plots the figure and provides a wt. mean age. For the manual entry, only a wt. mean age is provided, without a corresponding figure. Suggest adding in the option to plot that data on a figure in addition to calculating a wt. mean age from the manual entry.

jirhiker, 2019-09-13:
ok. I believe tho that if you do the manual entry, then hit save it should make the plot.

mjz1983, 2019-09-13:
OK. Didn’t know that option. WIll test.

Z

Matthew Zimmerer
Field Geologist II
New Mexico Bureau of Geology and Mineral Resources
New Mexico Tech
Socorro, NM 87801
575.835.5114
matthew.zimmerer@nmt.edu
https://geoinfo.nmt.edu/staff/zimmerer

mjz1983, 2019-11-25:
Option not working. CSV calc automatically updates when adding rows. However, when saving (in order to make plot) and error occurs.

mjz1983, 2019-11-25:
Didn't mean to close...

mjz1983, 2019-11-26:
Calculate is not working after the last update.

mjz1983, 2019-12-02:
CSV ideo is now plotting the ideogram. However, it no longer has the "calculate" button. Maybe this is no longer needed.

jirhiker, 2019-12-02:
correct. I removed the calculate button. Calculations should happen automatically

closed 2019-12-02 by jirhiker