NMGRL/pychron #1837: Error when saving CSV plot

nmgrluser, 2019-11-25:
active branch=develop

active analyses=

description=""

Traceback
```
Traceback (most recent call last):
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/pyface/ui/qt4/action/action_item.py", line 371, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/pyface/action/listening_action.py", line 74, in perform
    method()
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/tasks/task.py", line 348, in run
    self._run_pipeline()
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/tasks/task.py", line 535, in _run_pipeline
    self._run('run pipeline', 'run_pipeline')
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/tasks/task.py", line 514, in _run
    if not getattr(self.engine, func)():
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/engine.py", line 720, in run_pipeline
    if not node.pre_run(state, configure=configure):
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/nodes/base.py", line 109, in pre_run
    if self.configure(refresh=False, pre_run=True):
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/nodes/data.py", line 166, in configure
    self.unknowns = dsf.as_analyses()
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/csv_dataset_factory.py", line 186, in as_analyses
    ret = [FileAnalysis.from_csv_record(ri) for ri in self.records]
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/csv_dataset_factory.py", line 186, in <listcomp>
    ret = [FileAnalysis.from_csv_record(ri) for ri in self.records]
  File "/Users/argonlab2/.pychron.1/updates/pychron/processing/analyses/file_analysis.py", line 44, in from_csv_record
    setattr(obj, a, getattr(ri, a))
AttributeError: 'CSVRecord' object has no attribute 'k39'

```

mjz1983, 2019-11-25:
Still not working, regardless of whether using "Save" or "Save As".

jirhiker, 2019-11-25:
please retest, i had yet to push the updates prior to your last test

mjz1983, 2019-11-26:
"Save" and "Save As" still not working, although the error occurs at a different point in the process. Prior attempt I received an error message after clicking save. Now the error message is received after clicking save and then designating a location for the file.

closed 2020-01-13 by jirhiker