NMGRL/pychron #1840: Ideogram Error not updating

mjz1983, 2019-11-27:
When toggling the four different error options for Uncertainty Calculations for ideogram (e.g., include decay constant), the weighted mean error on the ideogram plot does not update. This is true for all four Uncertainty options (Positions error, include in mean, include irradiation error, include decay error).

jirhiker, 2019-11-28:
after some cursory testing I agree on` include decay error` and `include irradiation error`.

`Position error` may or may not work but it's only going to work if the Js were calculated with a position error. 

`Include in mean` was working for me

jirhiker, 2019-12-02:
fixed `include_decay_error` you should verify the uncertainty is being incorporated correctly tho

mjz1983, 2019-12-02:
What is the "correct way" to incorporate the decay constant?

jirhiker, 2019-12-02:
idk

closed 2022-01-24 by jirhiker