NMGRL/pychron #1862: Link outlier tag to plot type

mjz1983, 2019-12-18:
Suggest linking outlier tag status to plot type. For example, if steps on an isochron are tagged as outliers, that outlier status is then applied to the spectrum. But that information might be useful for integrated ages or something else. Likewise outliers of an ideogram might be useful on an isochron.

jirhiker, 2019-12-18:
not sure what you mean. Analyses tagged as omit/outlier/skip are omitted from an ideogram, spectrum, or isochron.

mjz1983, 2019-12-18:
Let me try again. The current configuration of tagging analysis is that if a data point is tagged as an outlier on one type of plot (e.g., isochron), it is tagged as an outlier on all plots (e.g., spectrum and/or ideogram) and in all calculations (e.g., integrated age)? 

An first alternative is to have users tag outliers for each plot. So for example, if you tag an analysis as an outlier on an ideogram it is not tagged as an outlier on the isochron, unless you specifically tag the analysis/set the status on isochron. 

The second alternative, similar to Mass Spec (forgive me...), is that you have the option of whether or not you want use the status from database when plotting data.

Perhaps unique to this user, but commonly an outlier on one plot isn't necessarily an outlier on another plot (e.g., outlier on an isochron vs an outlier in the integrated age calculation). Maybe this function currently exists and I am not using pychron correctly.

jirhiker, 2019-12-18:
ok. that's more what i thought. This was had already been an early aspect of tagging. you could tag analysis an define which plots that tag applied to.  No one ever used this functionality and it was too cumbersome to maintain. 

i can easily add the ability to ignore the database tags ala Mass Spec style

jirhiker, 2019-12-18:
actually the mass spec style was already available for ideogram's. Aka "Omit by Tag"

closed 2021-06-10 by jirhiker