NMGRL/pychron #1877: Flux problem with NM292C

mjz1983, 2020-01-29:
When fitting the flux for NM292C, I get this error message. Seems specific to this irradiation level. No problem fitting flux to other levels from NM292.

'''''

Traceback (most recent call last):
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/pyface/ui/qt4/action/action_item.py", line 371, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/pyface/action/listening_action.py", line 74, in perform
    method()
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/tasks/task.py", line 348, in run
    self._run_pipeline()
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/tasks/task.py", line 535, in _run_pipeline
    self._run('run pipeline', 'run_pipeline')
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/tasks/task.py", line 514, in _run
    if not getattr(self.engine, func)():
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/engine.py", line 728, in run_pipeline
    node.run(state)
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/nodes/find.py", line 285, in run
    for ip in ips if ip.identifier]
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/nodes/find.py", line 285, in <listcomp>
    for ip in ips if ip.identifier]
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/nodes/find.py", line 114, in _fp_factory
    pp = next((p for p in fluxes if p['identifier'] == identifier))
StopIteration
'''''

jirhiker, 2020-01-29:
NM-292C is corrupted or never had any samples/identifiers assigned to it

jirhiker, 2020-01-29:
looks corrupted

mjz1983, 2020-01-30:
Buzz Kill. Any chance to save it?

MZ

Matthew Zimmerer
Field Geologist II
New Mexico Bureau of Geology and Mineral Resources
New Mexico Tech
Socorro, NM 87801
575.835.5114
matthew.zimmerer@nmt.edu
https://geoinfo.nmt.edu/staff/zimmerer

closed 2020-02-03 by jirhiker