NMGRL/pychron #1910: JanAuto3SigBase bug

mjz1983, 2020-08-18:
When fitting isotopic evolutions for Jan, the JanAuto3SigBase does not include L2 or CDD fits. The baselines cannot be added and are not included in the drop down menu. Interestingly, if you select the Fit IsoEvo node in the Iso Evo pipeline without any populated analyses, the JanAuto3SigBase includes L2 and CDD fits, but these cannot be edited. See screenshots. Somewhat hard to explain. Hope this is clear.
[image]
[image]

jirhiker, 2020-08-18:
What is available in the drop down is based on the analyses. in the case where L2 and CDD don't show up what analyses are you selecting

mjz1983, 2020-08-19:
User error. Was plotting Felix data and accidentally used Jan set-u.

closed 2020-08-19 by mjz1983