NMGRL/pychron #1922: Unable to open Pipeline

cburrill, 2020-11-13:
active branch=dev/dr

active analyses=

description="When starting PyCrunch everything appears normal. However, when I go to VIEW and try to open the PIPELINE I get an error message and cannot open the PIPELINE."

Traceback
```
Traceback (most recent call last):
  File "C:\Users\Christine\anaconda3\envs\pychron3\lib\site-packages\pyface\ui\qt4\action\action_item.py", line 186, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "C:\Users\Christine\anaconda3\envs\pychron3\lib\site-packages\pyface\tasks\action\task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "C:\Users\Christine\.pychron.0\pychron\pychron\envisage\tasks\base_task.py", line 112, in perform
    application.open_task(self.task_id)
  File "C:\Users\Christine\.pychron.0\pychron\pychron\envisage\tasks\base_tasks_application.py", line 135, in open_task
    return self.get_task(tid, True, **kw)
  File "C:\Users\Christine\.pychron.0\pychron\pychron\envisage\tasks\base_tasks_application.py", line 124, in get_task
    win = self.create_window(w)
  File "C:\Users\Christine\anaconda3\envs\pychron3\lib\site-packages\envisage\ui\tasks\tasks_application.py", line 239, in create_window
    task = self.create_task(task_id)
  File "C:\Users\Christine\anaconda3\envs\pychron3\lib\site-packages\envisage\ui\tasks\tasks_application.py", line 191, in create_task
    task = factory.create_with_extensions(extensions)
  File "C:\Users\Christine\anaconda3\envs\pychron3\lib\site-packages\envisage\ui\tasks\task_factory.py", line 41, in create_with_extensions
    task = self.create(**traits)
  File "C:\Users\Christine\anaconda3\envs\pychron3\lib\site-packages\envisage\ui\tasks\task_factory.py", line 36, in create
    return self.factory(**traits)
  File "C:\Users\Christine\.pychron.0\pychron\pychron\pipeline\tasks\plugin.py", line 60, in _pipeline_factory
    from pychron.pipeline.tasks.task import PipelineTask
  File "C:\Users\Christine\.pychron.0\pychron\pychron\pipeline\tasks\task.py", line 34, in <module>
    from pychron.pipeline.engine import PipelineEngine, Pipeline, NodeGroup
  File "C:\Users\Christine\.pychron.0\pychron\pychron\pipeline\engine.py", line 38, in <module>
    from pychron.pipeline.nodes import FindReferencesNode, AuditNode
  File "C:\Users\Christine\.pychron.0\pychron\pychron\pipeline\nodes\__init__.py", line 33, in <module>
    from pychron.pipeline.nodes.figure import IdeogramNode, SpectrumNode, SeriesNode, InverseIsochronNode, \
  File "C:\Users\Christine\.pychron.0\pychron\pychron\pipeline\nodes\figure.py", line 34, in <module>
    from pychron.pipeline.plot.editors.map_figure_editor import MapFigureEditor
  File "C:\Users\Christine\.pychron.0\pychron\pychron\pipeline\plot\editors\map_figure_editor.py", line 18, in <module>
    import folium
ModuleNotFoundError: No module named 'folium'

```

jirhiker, 2020-11-13:
open a terminal.

enter the following 2 commands one after the other
```
conda activate pychron3
conda install folium

```

cburrill, 2020-11-13:
[image]

jirhiker, 2020-11-13:
sorry. try using 

conda activate pychron3  (you don't need to do this if the environment is already active e.g. if it the prompt says (pychron3) C:\Users\Cristine> )
pip install folium 

actually instead of doing that just try updating pychron.  I removed the "folium" dependency

cburrill, 2020-11-13:
Updating and restarting worked. I did get this message, not sure if it is important...
[image]

jirhiker, 2020-11-13:
nah don't need to worry about that message

closed 2021-02-09 by jirhiker