NMGRL/pychron #1944: Fitting Iso Evo to laser loads

mjz1983, 2021-03-10:
When fitting Iso Evo to laser loads C0226 and C0227, only a single type of analysis can get fit at one time. For example, instead of loading u, bu, c, bc, a, and ba and fitting them all at once, only unknowns can be selected then fit. Only blanks can be fit. Only cocktails can be fit. Only airs can be fit.

jirhiker, 2021-03-10:
You select which type of analyses are fit
[image]

mjz1983, 2021-03-10:
At least on ArgonLab2, even if you select blank unknown, cocktail, and unknowns, you can only fit unknowns rather than all 3 at one time.

mjz1983, 2021-03-10:
Could replicate on ArgonLab2. Another puzzle, when selecting all the unknowns from C0226, the data from NM312O are not displayed in the Iso Evo fit pipeline.

mjz1983, 2021-03-10:
Even more data. Every CO2 or diode tray that includes projects that needed an IC-update-fix are impacted. To the best of my knowledge that D0068 and 69, C0214, 226, and 227.

jirhiker, 2021-03-10:
the missing data issues should be resolved in the latest update. Or at least if data fails to load it does not pass silently like what is happening now. 

The current bug involving missing data from NM312O has to do with a bit of a catch-22. When loading the data pychron tries to get the flux history for the analysis, however no flux fitting has been done for 312O, but you cant do any flux fitting because you cant load the data.  

The bug preventing analyses with no flux history from being loaded has been resolved.

mjz1983, 2021-03-10:
Great. Thanks. We’ll test drive it tomorrow.

Z

Matt Zimmerer
Field Geologist II
New Mexico Bureau of Geology
New Mexico Tech
801 Leroy Pl
Socorro, NM 87801
Matthew.Zimmerer@nmt.edu
575.835.5114

closed 2021-03-30 by jirhiker