NMGRL/pychron #1946: Upcoming breakage in Enable 5.0.0+

jwiggins, 2021-03-16:
Hi there! I'm one of the maintainers for the Enable package which you're making use of in this project.

We've been doing some disruptive refactoring as part of 5.0.0 and later versions. I wanted to point out that a couple of your uses of `kiva.fonttools` will no longer work once you upgrade to Enable 5.0.0 (or later).

This import will no longer work, but it's also unneeded (see below):
https://github.com/NMGRL/pychron/blob/e49313146a89dbdf55eec30e097218374a3a441c/pychron/core/pdf/save_pdf_dialog.py#L22
https://github.com/NMGRL/pychron/blob/e49313146a89dbdf55eec30e097218374a3a441c/pychron/core/pdf/save_pdf_dialog.py#L38-L47
I suggest:
```python
from kiva.api import Font, NORMAL

for face in pychron_constants.TTF_FONTS:
    for face_name in (face, face.lower()):
        spec = Font(face_name=face_name, style=NORMAL, weight=NORMAL).findfont()
        try:
            tf = TTFont(face_name, spec.filename, subfontIndex=spec.face_index)
            pdfmetrics.registerFont(tf)
            pdfmetrics.registerTypeFace(TypeFace(face_name))
        except TTFError as e:
            print('invalid font', font, e)
```

Starting with 5.0.0 this will [most likely] fail (because `fontManager` is None):
https://github.com/NMGRL/pychron/blob/41f58470c78f05cb944b5dccb1943007aacfbbef/pychron/pychron_constants.py#L50-L55
We recommend against it, but if you want the `FontManager` instance, you should call `default_font_manager` from `kiva.fonttools.font_manager`. Further, starting in Enable 5.1.0, the `defaultFamily` attribute has been renamed to `default_family` and is a string instead of a dictionary. It's default value is "sans-serif" which will likely give the same result as what you're doing here.
That said, `FontManager` is an internal implementation detail and no guarantees are made about it's API stability.

If you have any more questions, don't hesitate to reach out. Thanks for using Enthought open source!

jirhiker, 2021-03-16:
@jwiggins Thank you very much. 

Ive been using ETS for quite a while now and it's been absolutely essential to the success of pychron.  What's the best way to start contributing to Enthought projects and start giving something back.

jwiggins, 2021-03-16:
> What's the best way to start contributing to Enthought projects and start giving something back.

One easy way to contribute would be opening issues for bugs you've run into. If you give good reproduction instructions or (even better) a minimal working example, there's a chance it will get fixed quickly.

Other than that, we do accept pull requests from external developers. Although if you're adding features it might be best to start a discussion via an issue first. We have a `good first issue` tag on several of our projects which we use to help interested developers get their feet wet.