NMGRL/pychron #1948: History in recall and Spectrum History not matching

mjz1983, 2021-03-16:
When viewing history of data reduction, this history via the recall window and history via the history node in the pipeline, the two histories do not match. Using example MW-031 L#66969-01. This data was 'worked up' in August 2019 and has not be worked until the IC update on Feb 8th, 2021. However, when viewing the history vis the History pipeline there appears to be ~25 additional history rows included refitting of isotope, blanks, IC factors which was done to this sample.
[image]

jirhiker, 2021-03-16:
The two lists of changes represent different things.

The list of changes in the recall window is just the changes specific to that analysis.

The list of changes in the History Node is all the changes to repository(ies) that contain(s) the selected analyses

mjz1983, 2021-03-16:
Making sure I understand. The history node is not specific to the sample, but to the repository. 

Let’s say I have two samples (A and B) in repository Z. I reduce A and B data at the same time thus have the sample history. But if I go back and refit the isotopes to B, the history node for A will also show a new fit to the isotopes. Is that correct?

Matt Zimmerer
Field Geologist II
New Mexico Bureau of Geology
New Mexico Tech
801 Leroy Pl
Socorro, NM 87801
***@***.***
575.835.5114

jirhiker, 2021-03-16:
History node is not specific to the sample (currently). Its all the changes to the selected repository

Let’s say I have two samples (A and B) in repository Z. I reduce A and B data at the same time thus have the **SAME** history. But if I go back and refit the isotopes to B, the history node for **repository Z** will also show a new fit to the isotopes.

jirhiker, 2021-03-16:
In the History Node I could see the benefit of only displaying the changes for the selected analyses. Ill look into that

mjz1983, 2021-03-16:
Personally, I think only seeing the changes in the history to the sample, rather than the entire repository is more beneficial.

Z

Matt Zimmerer
Field Geologist II
New Mexico Bureau of Geology
New Mexico Tech
801 Leroy Pl
Socorro, NM 87801
***@***.***
575.835.5114

closed 2021-03-30 by jirhiker