NMGRL/pychron #1967: Add ability to view a view linked with an analysis.

jirhiker, 2021-04-08:
Add ability to view a view linked with an analysis. 

From a recall tab select "Find Video" in the toolbar.  This action should only be active (or visible) when MediaStorage Plugin is enabled and a recall tab is selected. 

This action will then use the runid and load number to locate the file. 
     Maybe pyexperiment should save to the MediaTbl when it records a video?

Another entry point is right click in the Analysis pane, "Play Video". This would allow multiple videos to be played side by side potentially. 

if a video is located then a video pop up is opened. 
Controls include, play, stop, speed, frame counter, frame advance

jirhiker, 2021-04-09:
implementing a robust video player in pychron going to be to difficult currently. 

as a first step, "Find Video" simply will show the path to the video if available. Maybe it opens it in VLC

jirhiker, 2021-04-09:
have a potential working video player using vlc. 

need to install using pip install python-vlc

mjz1983, 2021-06-09:
Seems to be working. Z

closed 2021-06-10 by jirhiker