NMGRL/pychron #1986: Fix outlier symbols

mjz1983, 2021-05-25:
Outliers on ideograms are not currently displayed correctly. Older version of pychron displayed ideogram outliers as a hollow circle (black outline, white center). Now outliers appear as a white ouline and black interior, which is nearly impossible to see. See screenshot.
[image]

Is outlier appearance  even an option in the ideogram appearance?

mjz1983, 2021-05-25:
Correction, the data points with white outlines and black interiors are subgrouped analyses, not outliers. There doesn't seem to be a way to change the presentation of one type of data (SCLF) vs another (subgrouped integrated age of SCSH). A bug on your end?

jirhiker, 2021-05-25:
Omitted points are always displayed as white squares with the group color as an outline. 

Subgrouped ages are now displayed with an outline color that is the inverse of the group color.  There currently is no control over this appearance so AFAIK the work arounds would be use a non-white background or don't use black as a group color.

mjz1983, 2021-05-25:
When omitting a subgrouped analysis the user can't see the change. Almost all ideograms have a white background, particularly for appendices. And black color group is the standard for plotting a single group.

Why change in the subgrouped analysis display. It seemed fine the way it way. At the least, can this be an option to display subgrouped analyses the same as regular data.

jirhiker, 2021-05-25:
"Why change in the subgrouped analysis display. It seemed fine the way it way. At the least, can this be an option to display subgrouped analyses the same as regular data." It was explicitly asked for.

Ill add the ability to turn off

closed 2021-06-09 by mjz1983