NMGRL/pychron #1990: Isochron values change when rescaling x-axis

mjz1983, 2021-06-03:
When rescaling the x-axis of an inverse isochron, the 40/36 intercept, 40/39 intercept (age), and MSWD values change even though the points included in the calculation remain the same.

See attached screen shot of VC-1 (n=35 points).
[image]
[image]

jirhiker, 2021-06-03:
ok. something is happening with the omitted points

The omitted points are different between the 2. Are the omitted points in the first figure tagged as omitted/outliler or did you omit those points manually? or is a mix?  

Do you see that same behavior with no omitted points? with no tagged omitted points? with points tagged as omit but manually included?

mjz1983, 2021-06-03:
Omitted points are the same between the two. Points are omitted by tagging. Seeing the same behavior when all points are included.

Noticing new odd behavior in that the isochron is not plotting points according to tagged status.

mjz1983, 2021-06-03:
Pychron appears to be randomly tagging points from the Analyses window. For example, aliquot -04 is getting removed from the isochron regression even though it is not tagged as an outlier
[image]

jirhiker, 2021-06-04:
k thanks. its definitely getting confused as to which points to treat as omitted