NMGRL/pychron #1993: Auto Push

mjz1983, 2021-06-09:
Suggest adding Auto Push to compliment Auto Pull. Users can opt out of Auto Push if the like. This request stems from the fact that student projects often have at least two two workers (the student and the advisor), commonly using three computers (student's computer, advisor computer, and ArgonLab2). Auto Push would help with assessing the status of the data (e.g., have the iso evo been fitted, or were they fitted but not pushed).

jirhiker, 2021-06-09:
Check out the latest version. "Fit" pipelines now end with a Push Node.  You can make it auto push by checking "Auto Push" in Preferences/DVC, (If Auto Push is unchecked user is still asked to push/share data when the pipeline completes, in addition to begin asked when the pipeline window closes)

mjz1983, 2021-06-09:
Beautiful. Next to the linked video, this might be my new favorite feature. Will test tomorrow.

Z

closed 2021-07-13 by nmgrluser