NMGRL/pychron #1995: sigma symbol missing on ideogram

mjz1983, 2021-06-17:
Missing sigma symbol on ideogram. Sigma symbol is replaced with a square box. This requires users to open ideogram plots in adobe illustrator and add in the sigma symbol manually.

[image]

jirhiker, 2021-06-17:
see #1358

this is not reproducible on all computers. Works on my computer
[image]

mjz1983, 2021-06-17:
Time for a new ArgonLab2 comp?

Matt Zimmerer
Field Geologist II
New Mexico Bureau of Geology
New Mexico Tech
801 Leroy Pl
Socorro, NM 87801
***@***.***
575.835.5114