NMGRL/pychron #2017: Add Min Sep Notes in recall

mjz1983, 2021-09-21:
Suggest adding mineral separation notes to recall window. Maybe add it to the history window?

jirhiker, 2021-10-01:
very few samples have sample prep comments. What "notes" do you want to actually display in the recall window.

mjz1983, 2021-10-03:
I guess this is more sample notes, rather than min sep notes. When entering samples I sometimes include notes such as melt-inclusion rich, picked in oil, very fine grained crystals, etched by HF, compare to sample XXXX, etc…

I forget these comments in the 6 months between prep and dating. So I was thinking it would be nice to see any comments like that in the recall window, but perhaps there is a better was to retrieve those notes, if any.

Matt Zimmerer
Field Geologist II
New Mexico Bureau of Geology
New Mexico Tech
801 Leroy Pl
Socorro, NM 87801
***@***.***
575.835.5114

jirhiker, 2021-10-06:
added. 
 But I suspect this feature will be less useful than anticipated.

For example you have 2 samples in the database named '17L-15' One for sanidine, one for Sanidine 60-20. You added the note "20% crystals plag>>qtz, abund. san, bt" to '17L-15 Sanidine 60-20'  but no note to '17L-15 Sanidine'.  When the sample was entered in the irradiation the sample '17L-15 Sanidine' was chosen.  Therefore the analysis is tied to sample '17L-15 Sanidine' and when the analysis is recalled no sample note is available.   PizzaTracker and Entry/Sample allow you to  recall the notes for any sample much easier

closed 2022-01-24 by jirhiker