NMGRL/pychron #2065: Spectrum not plotting correctly

mjz1983, 2022-01-03:
When plotting a spectrum, the plot is "squished" vertically. Changing the vertical or horizontal fixed width doesn't appear to updating plot.
[image]

mjz1983, 2022-01-03:
Note, this became a project after the last updates on ~Dec. 23rd.

jirhiker, 2022-01-03:
[image]
[image]
[image]

mjz1983, 2022-01-03:
Suggests this might be a problem specific to ArgonLab2 (where I noticed the problem). We saw this on the newest lab computer and I think you though it was related to too many version on a computer.

???
Matt Zimmerer
Field Geologist II
New Mexico Bureau of Geology
New Mexico Tech
801 Leroy Pl
Socorro, NM 87801
***@***.***
575.835.5114

jirhiker, 2022-01-05:
The issue appears when fixed_width and fixed_height are set to 0.  
a work around is to set fixed_width to a positive integer.

jirhiker, 2022-01-05:
on closer inspection this is not a bug.

If fixed_width and fixed_height are not set then pychron respects the Height setting for each plot.  in the above example all the plots are set to a height of 100

closed 2022-01-05 by jirhiker