NMGRL/pychron #2102: New Argon lab computer not pushing data on pychron

mjz1983, 2022-01-24:
Auto pushing of data does not appear to be functioning on the new Argon Lab computer. Data for D0098 and 0099, reduced on this computer, is not updated on ArgonLab2 (no history of data reduction is shown in the recall window)

mjz1983, 2022-01-24:
After further inspecttion, this might only pertain to the East San Juan Conjeos project in loads 098 and 099. All other projects/repositories in this two loads appears to be updated on both lab computers. Repositores>pull data does not appear to update the East San Juan Conjeos repositores. Suggestions on how to get updated dated on ArgonLab 2?

jirhiker, 2022-01-24:
14 commits were not pushed on Argonlab3

closed 2022-01-24 by jirhiker

mjz1983, 2022-01-24:
Doesn't auto push in pychron preferences take care of that?

jirhiker, 2022-01-26:
For whatever reason the changes were not automatically pushed. It would take a lot more investigation to discover why during this specific data reduction session the changes where not pushed (perhaps the pychron tried to auto push but did not have a connection to github because the internet was down, or github was down, or the bureau network was down, etc. Or perhaps there was some issue with this specific data preventing pychron from auto pushing, but this is unlikely as i didn't have to fix anything when I manually pushed)

Because auto push cannot be relied on 100% (i.e. if the internet goes down just as you try to push) the Repository view exists so that you can interrogate the status of the relevant repository.  

Also all of the pushing/pulling/repository syncing can be done completely independently of pychron; one can use `git` directly

I can add a user warning if auto pushing fails so that its less of a mystery