NMGRL/pychron #2104: outlier status incorrectly applied to integrated age

mjz1983, 2022-01-25:
When making data tables, outlier status (in the attached example, outlier on isochron) is being applied to the total integrated age. The total integrated age should include all steps.
[SMCN-01 hbl table.xlsx](https://github.com/NMGRL/pychron/files/7936306/SMCN-01.hbl.table.xlsx)

jirhiker, 2022-01-25:
This is a feature. you specify how the total integrated age should be calculated, all steps or all valid steps

mjz1983, 2022-01-25:
Ah. I might need a quick hand to show me how to turn on/off this feature.

Matt Zimmerer
Field Geologist II
New Mexico Bureau of Geology
New Mexico Tech
801 Leroy Pl
Socorro, NM 87801
***@***.***
575.835.5114

mjz1983, 2022-01-25:
When the outlier steps are included in the total integrated age, those
outlier steps are also used in the isochron calculation, which they
shouldn't.

Is it possible to generate a data table where steps are omitted on the
isochron, but include in the total integrated age?

MZ

jirhiker, 2022-01-25:
The expected behavior is 

- Total Integrated will use all steps whether they are omitted or not. 
- Valid Integrated uses only the non-omitted steps

cburrill, 2022-02-02:
For the tables I think what we need is an option (check box) to use either the Total Integrated or Valid Integrated Age. Currently under the summary rows section, there is only the option for the Total Integrated age. So if steps are marked invalid and they are included on the table, they are currently being used in the integrated age and you need to change the integrated age manually or not include the invalid steps at all. I think the preference would be to show the invalid steps on the table but not have them included in any of the age calculations.

[image]

[image]

[image]

jirhiker, 2022-02-02:
Don't know if this addresses your issue but the preferred age kind (Total Integrated, Valid Integrated) is set at the Group Age Node. 

[image]

cburrill, 2022-02-03:
No, the correct option to change it isn't there. The age row changes whats in the summary where the plateau age goes. Even though I have "plateau else valid integrated age" in all the rows, it still shows "total integrated age" in the summary.

[image]