NMGRL/pychron #2106: Table Issues (formatting and missing data)

cburrill, 2022-01-26:
**Created a table using "Subgrouped Analyses" pipeline and ran into the following issues:**

1. The plateau information shows up in the area with the step heating analyses data (highlighted yellow in file)

2. The border line underneath the heating steps (row 18 in this case) no longer goes under all the columns but there is now a line that does goes under all columns above the notes. Preference would be to have only the line under the heating steps that appears under all columns, if possible.

3. In the Unknowns tab of the resulting spreadsheet, there is a column (BW) with header "(38Ar/38Ar)K" but I think it should be (38Ar/39Ar)K and the column is empty as well as the error column next to it. (outlined in red box in file)

**To Reproduce**
See screenshots of XLS options

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Most up-to-date version of PyCrunch (as of 1/25/22)

[MW-032-02.xlsx](https://github.com/NMGRL/pychron/files/7938053/MW-032-02.xlsx)
[image]
[image]

jirhiker, 2022-01-26:
k. thanks. those seem straightforward to fix

jirhiker, 2022-01-26:
1. I believe you made a subgrouped table (?). Therefore the display of the plateau information with the steps is by design. its showing the age for that subgroup.

jirhiker, 2022-01-26:
Great. Problem 3 should be addressed in the latest version

closed 2022-02-01 by jirhiker