NMGRL/pychron #2126: CSV comparison still not working

TCan13, 2022-02-09:
I put the data into pychron itself and it seemed to work fine then once the ideogram should have came up it gave an error message and a blank screen. I attached the terminal screen right after that happened.

[image]
[image]

jirhiker, 2022-02-09:
you probably have an aux plot selected in the Ideogram options. For example since you didn't enter a %40rad value in the csv you can't make a %40Rad plot.  make sure only ideogram and analysis number are selected plots

TCan13, 2022-02-09:
That makes sense. Got it working now, thanks.