NMGRL/pychron #2148: Inconsistency between spacing between symbols and text

mjz1983, 2022-02-25:
On nearly all the plots, there are inconsistencies w.r.t to spacing between symbols and text.

Example of an ideogram is attached.

There are spaces before and after the ± symbol. No spaces between MSWD, equal symbol, and the MSWD value. No space between n and the equal sign, but a space after the equal sign and number of analyses used.

[image]

jirhiker, 2022-02-25:
There has been no consensus or any indication of what people want.  Identify the format you want and provide specific examples where the format is not followed

mjz1983, 2022-02-25:
I suggest we follow GSA style, which appears to include spaces before and
after symbols (e.g., 20.10 ± 0.02 Ma, MSWD = 2.22, n = 13/13). Recent pubs
by Matt, Bill, and myself in a variety of journals also follow a
similar style.

Right now, I don't think any of the plots (spectrum, ideogram, or isochron)
follow this pattern, nor do they follow any consistent pattern (variable
space before and after symbols).

closed 2022-03-23 by jirhiker