NMGRL/pychron #2190: Failed to Sync repository

nelsiverson, 2023-01-04:
I am not able to push changes to the Morgan01099 repository from my laptop. 
Preferences<DVC< says that I am connected. 

Below is the log error

Repositories                            : 2023-01-04 13:58:41,044 DEBUG     (MainThread) find changes. names=('Morgan01099<GitRepo>',)
DVC                                     : 2023-01-04 13:58:46,645 DEBUG     (MainThread) push repository <pychron.git_archive.repo_manager.GitRepoManager object at 0x00000216BBF90648>
DVC                                     : 2023-01-04 13:58:46,645 DEBUG     (MainThread) pushing to remote=origin, url=https://github.com
GitRepoManager                          : 2023-01-04 13:58:47,263 DEBUG     (MainThread) Traceback (most recent call last):
  File "C:\Users\Nels\.pychron.0\pychron\pychron\git_archive\repo_manager.py", line 819, in push
    self._repo.git.push(remote, branch)
  File "c:\Users\Nels\.conda\envs\pychron3\lib\site-packages\git\cmd.py", line 542, in <lambda>
    return lambda *args, **kwargs: self._call_process(name, *args, **kwargs)
  File "c:\Users\Nels\.conda\envs\pychron3\lib\site-packages\git\cmd.py", line 1005, in _call_process
    return self.execute(call, **exec_kwargs)
  File "c:\Users\Nels\.conda\envs\pychron3\lib\site-packages\git\cmd.py", line 822, in execute
    raise GitCommandError(command, status, stderr_value, stdout_value)
git.exc.GitCommandError: Cmd('git') failed due to: exit code(128)
  cmdline: git push origin master
  stderr: 'remote: Invalid username or password.
fatal: Authentication failed for 'https://github.com/NMGRLData/Morgan01099.git/''

GitRepoManager                          : 2023-01-04 13:58:47,263 WARNING   (MainThread) Morgan01099<GitRepo> push failed. See log file for more details
Repositories                            : 2023-01-04 13:58:49,909 DEBUG     (MainThread) find changes. names=('Morgan01099<GitRepo>',)

jirhiker, 2023-01-04:
Are you using the latest version?  If so try reseting your github authentication manually by deleting the file

`Pychron/.appdata/oauth.json`

nelsiverson, 2023-01-04:
Fixed! Thanks

closed 2023-01-04 by jirhiker