NMGRL/pychron #2204: Unable to save pdf of ideogram

mjz1983, 2023-03-24:
Error message:

Traceback (most recent call last):
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/reportlab/pdfbase/pdfmetrics.py", line 711, in getFont
    return _fonts[fontName]
KeyError: 'helvetica'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/pyface/ui/qt4/action/action_item.py", line 371, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/pyface/action/listening_action.py", line 74, in perform
    method()
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/tasks/task.py", line 457, in save_figure_pdf
    save_pdf(ed.component, path=path, options=sfm.pdf_options, view=True)
  File "/Users/argonlab2/.pychron.1/updates/pychron/core/pdf/save_pdf_dialog.py", line 150, in save_pdf
    component, valign=options.valign, halign=options.halign
  File "/Users/argonlab2/.pychron.1/updates/pychron/core/pdf/pdf_graphics_context.py", line 220, in render_component
    component.draw(self, view_bounds=(0, 0, width, height))
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/enable/component.py", line 427, in draw
    self._draw(gc, view_bounds, mode)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/enable/component.py", line 779, in _draw
    self._dispatch_draw(layer, gc, view_bounds, mode)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/enable/container.py", line 273, in _dispatch_draw
    component._dispatch_draw(layer, gc, new_bounds, mode)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/enable/container.py", line 273, in _dispatch_draw
    component._dispatch_draw(layer, gc, new_bounds, mode)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/enable/container.py", line 273, in _dispatch_draw
    component._dispatch_draw(layer, gc, new_bounds, mode)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/enable/component.py", line 799, in _dispatch_draw
    handler(gc, view_bounds, mode)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/enable/component.py", line 877, in _draw_overlay
    overlay.overlay(self, gc, view_bounds, mode)
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/plot/overlays/mean_indicator_overlay.py", line 292, in overlay
    o.overlay(other_component, gc, view_bounds=view_bounds, mode=mode)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/chaco/plot_label.py", line 89, in overlay
    self._draw_overlay(gc, view_bounds, mode)
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/plot/overlays/mean_indicator_overlay.py", line 97, in _draw_overlay
    self._label.draw(gc)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/chaco/label.py", line 169, in draw
    gc.set_font(self.font)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/kiva/pdf.py", line 629, in set_font
    self.gc.setFont(face_name, font.size)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/reportlab/pdfgen/canvas.py", line 1691, in setFont
    font = pdfmetrics.getFont(self._fontname)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/reportlab/pdfbase/pdfmetrics.py", line 713, in getFont
    return findFontAndRegister(fontName)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/reportlab/pdfbase/pdfmetrics.py", line 695, in findFontAndRegister
    face = getTypeFace(fontName)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.7/site-packages/reportlab/pdfbase/pdfmetrics.py", line 652, in getTypeFace
    return _typefaces[faceName]
KeyError: 'helvetica'

mjz1983, 2023-03-24:
The font Arial doesn’t work either. Why does helvetica work for isochron and spectrum on ArgonLab2? 

Matt Zimmerer
Interim Mapping Program Manager
Field Geologist II
New Mexico Bureau of Geology
New Mexico Tech
801 Leroy Pl
Socorro, NM 87801
***@***.***
575.835.5114

mjz1983, 2023-03-24:
The problem appears to be with ArgonLab2. However, ideograms cannot be saved with any of the fonts whereas all fonts work with spectrum/isochron.  Does this mean ArgonLab2 can no longer be used for making pdfs of ideograms?

jirhiker, 2023-03-25:
Thanks I think that should be useful information. 

I'll have to get back to you on argonlab2. I don't know its status

jirhiker, 2023-03-25:
I think there is a specific element in the ideogram that is breaking things. If you turn off mean indicator that my work

mjz1983, 2023-03-27:
Turning off mean indicator fixes the problem. Of course an ideogram without that info isn’t super helpful.

Matt Zimmerer
Interim Mapping Program Manager
Field Geologist II
New Mexico Bureau of Geology
New Mexico Tech
801 Leroy Pl
Socorro, NM 87801
***@***.***
575.835.5114